maxieds/ChameleonMiniLiveDebugger #1: Testing ChameleonMiniLiveDebugger

gtpy, 2018-01-06:
I’ll have done some first checks of your app.
It looks nice.
Some little things can be added:

Similar functions like in a serial terminal:
[image]

f.e. change settings, query actual setting, query actual config, ...
Would be great if you can add some like these functions.
(Here some example commands from my mobile terminal)

Why you only have under configure minimal peripherals only rbuttonlong and not the lbutton, lbuttonlong,...

Cheers

maxieds, 2018-01-08:
Ok, I will work on adding some of your suggestions. I have comp exams this week and next which are taking precedence over all for now. So it might be a little while before I get back into coding. 

I just added the two so as not to clutter up the already large tools menu interface too much. It’d easy to add them all. I’ll put it on my todo list.

What do you think about the current APDU command recognition? Is it accurate enough, or still needs some improvement?

gtpy, 2018-01-09:
Hi @maxieds 
I've performed some small tests.
Your app crashes when screen rotates - so i need to turn off screen rotation before usage.
Logged 3 different connections and like the output of your log-export.

maxieds, 2018-01-09:
Would you add the screen rotation bug to the issues page on my GitHub site
for the app so I remember to fix it? I'm sure I can have the app turn this
off by default so you don't have to.

Are you finding the apdu command recognition to be accurate, or do you
think it still needs work?

Maxie

maxieds, 2018-01-09:
Sorry, I guess you already did this. I was expecting it to be from the CM issues site on my email. :)

maxieds, 2018-01-09:
The screen rotation issue is fixed in the development version. I will roll out a new release on the Play Store later tonight (US Eastern time).

gtpy, 2018-01-09:
Thanks. I'll test deeper if I find more time between work and family.
I think to find some time at work to play around with the chameleon and your app
I'll give you an update when I looked at apdu command and more ;-)

maxieds, 2018-01-10:
The [newest live version in the Play Store](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) should incorporate several of your suggestions. It also fixes the screen rotation issue in the software without the user having to to it. Let me know what you think when you have some more time to test it out. 

@doegox: I noticed that you starred the repo and are located in Brussels, so I added a roll out in Belgium for you to use without having to proxy through an install. Please also give me your feed back on the new release.

doegox, 2018-01-10:
Excellent thank you @maxieds !
I tried briefly with a OnePlus 5, a USB-C OTG and the usual USB micro cable and it can communicate properly with the Chameleon Mini (once OTG storage is activated in the parameters of course).
The few issues I had yesterday seem to be already solved in today version, nice :) (log tab empty, second instance launched on re-plug/reset,...)

doegox, 2018-01-10:
How do we see the result of a command? E.g. using the "(Almost) Complete Shell / LEDRED?" I see in the log tab "INFO: shell command of ledred? returned status 101:ok with text" but I don't see the expected text.
Also when I press the predefined command buttons (charging, identify,...) I don't see any effect, no result, nothing in the log tab.

doegox, 2018-01-10:
FYI for those having a phone with USB-C, I received today this cable https://www.amazon.fr/gp/product/B071H6ZW6Y and it works very nicely.

maxieds, 2018-01-10:
The output of the command should be directly underneath the header (if
there is any to display). I found a couple more bugs, so I will release a
newer version this morning. I'm glad you're enjoying the newer release.

I still don't know how to fix the blank tabs when the app relaunches
because usb is plugged in. Something weird is going on with the way the USB
intent handlers are getting registered. It's on my list of bugs to fix, but
for now if it happens just restart the app.

Maxie

maxieds, 2018-01-10:
A new release (0.0.8-v1) should be out soon. It features improved APDU command recognition and some bug fixes. Thanks for helping me test it!

doegox, 2018-01-10:
on the new version (0.0.8) peripherals and log mode, we don't see anymore the initial status of all leds & buttons. It was properly displayed in 0.0.6

maxieds, 2018-01-10:
I'm going to have to fix this later this afternoon. Let me know if the new
release broke anything else.

Maxie

maxieds, 2018-01-11:
Here's a more recent screen shot of what the app looks like now. Thanks to @doegox for all of the suggestions. I'm going to roll out another release on play store once I figure out how to get this really annoying blank tabs after USB connect issue fixed. Look out for it tomorrow morning. 
[image]

maxieds, 2018-01-15:
I'm going to close the issue since it seems that all of the original requests have been handled. Please post new issue threads for future feature requests in the app. I'd like to see more of them.

closed 2018-01-15 by maxieds