maxieds/ChameleonMiniLiveDebugger #2: New Feature Requests

maxieds, 2018-01-09:
Here are some features I want to add:

- Better APDU command recognition
- Ability to download a log binary file from the device (LOGDOWNLOAD) using a XModem connection and display it as though it were a live log
- Query the device for its current settings of the peripherals (_working on this now_)

maxieds, 2018-01-11:
# Add Detailed JavaDocs (This is important)

... And will get done eventually. It's just my least favorite part of the software engineering scheme of things. To much other cool stuff to implement and get working first.

maxieds, 2018-01-15:
The primary "mainstream" (i.e., built-in to the Chameleon Mini command line) feature that still needs to be added at this point is the uploading of card data via XModem.

closed 2018-01-15 by maxieds

jankais3r, 2018-07-31:
It would be awesome to have a "1-button-cloning" in the app. Something similar to [this](https://github.com/emsec/ChameleonMini/pull/103/commits/efe1c5d89924f90a997700b8ee8d80096164dc0f).

maxieds, 2018-07-31:
I just barely skimmed the source code you linked (I'm super busy today!). So what you want is a 1-button way to clone an external tag to the Chameleon Mini, is that correct?