maxieds/ChameleonMiniLiveDebugger #4: App Screenshots

maxieds, 2018-01-15:
Mostly to get GitHub to host the screenshots for the page instead of linking to the Play Store images, which tend to change with new releases, here are several screenshots of the running app and its output: 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]

maxieds, 2018-01-15:
Anyone else have any useful screenshots and/or app usages to share?

gtpy, 2018-01-15:
@maxieds I’ll test your app today or tomorrow. Then I’ll post some screenshots from usage.

maxieds, 2018-01-15:
New Tools Menu shot with the working threshold slider:
[image]

gtpy, 2018-01-17:
Here are some more screenshots of the usage from v0.2.8

**Overview - With loaded slot#1 (mf_ultralight):**
[image]

**Print actual version of ChameleonMini:**
[image]

**Reader Mode - slot#4 - identify (mf_ultralight) command:**
[image]

**Reader Mode - slot#4 - identify (mf_classic_1k) command:**
[image]

**Peripherals and Log Mode -> changed LBS (left button short) = UID_RANDOM:**
(only changed middle RND, cause it's a modified firmware which I am running - original FW will randomize full UID)
[image]

**Changed UID RANDOM#2:**
[image]

maxieds, 2018-01-17:
Very cool. Do you have any interest in the themed flavors?

One other feature that might be useful would be to dump an actual mfu card
into a format that can be uploaded as a card image. Any idea how difficult
that would be?

<https://user-images.githubusercontent.com/13873777/35063582-cedd74d2-fbc7-11e7-81a5-0cfc5e3df6bf.png>

<https://user-images.githubusercontent.com/13873777/35063665-0b4623ce-fbc8-11e7-9b29-d65ed693ecbf.png>

<https://user-images.githubusercontent.com/13873777/35063727-43856402-fbc8-11e7-9c2d-7074eb45285f.png>

<https://user-images.githubusercontent.com/13873777/35063766-5ce6b162-fbc8-11e7-87e3-228a1bcf86a2.png>

<https://user-images.githubusercontent.com/13873777/35063838-8e262ae6-fbc8-11e7-81a5-a1b66a9808ef.png>

<https://user-images.githubusercontent.com/13873777/35063865-a11b0856-fbc8-11e7-9ac7-629517799b61.png>

gtpy, 2018-01-18:
Sorry there was a little error in my compiled FW
Now the new build works like charm and shows the UID and IDENTIFY correct in header
[image]

closed 2018-01-26 by maxieds

maxieds, 2020-09-09:
[image]

maxieds, 2020-09-09:
[image]
[image]

maxieds, 2020-09-09:
[image]

maxieds, 2020-10-14:
Treating this long gone issue as an image upload thread for recent screenshots to be added soon to the WIKI...

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

maxieds, 2020-10-14:
[image]

maxieds, 2020-10-14:
[image]
[image]
[image]

maxieds, 2020-10-14:
[image]

maxieds, 2020-10-14:
[image]

maxieds, 2020-10-15:
[image]

maxieds, 2020-10-20:
[image]
[image]

maxieds, 2020-10-20:
[image]

maxieds, 2020-11-18:
[image]

maxieds, 2021-01-12:
[image]
[image]