maxieds/ChameleonMiniLiveDebugger #5: How to use misc configuration (advanced Chameleon shell) v0.1.9

gtpy, 2018-01-15:
@maxieds 
How to use the advanced settings under misc configuration ?
For example if I select SETTING=
Need I to use the input field above? But both doesn’t give me a change of the slot.

I get error: Command formatting error: the input user bytes are invalid or not of the correct length 

Maybe I miss something...

maxieds, 2018-01-15:
This is what I used to change the slot:
[image]

Does it work for you now?

BTW: When you start testing the LOGDOWNLOAD and DOWNLOAD features in the Export tab, please make sure you are using the most recent v0.2.1 version (on Play Store, should be out momentarily). I caught a bug that is now fixed and updated the baud rate on the connection _significantly_ so the downloads really fly now.

maxieds, 2018-01-15:
And here's how to change the UID (must be the correct number of bytes for the card selected):
[image]

gtpy, 2018-01-15:
Ok. That work. 
The failure before was, because i pressed enter after typing in the slot number.
So i didn't see that the entered number is still in the field but i couldn't see it, because it jumped into new line.

If I use this field in combination with changing the slot it stay at SETTING= 
Now if i want to change to another slot i need to select in drop down --Advanced Chameleon Shell-- and back to SETTING= to change again.

maxieds, 2018-01-15:
I think I fixed the latter behavior. Try [this (working development) APK](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/download/v0.2.0/app-release.apk).

maxieds, 2018-01-15:
There's really not a graceful solution to this without defining a custom Spinner class in Android. This fix just resets the position to the "Advanced Shell" option each time.

gtpy, 2018-01-15:
This solution works.
It’s much easier to change between slots.
I’ll test more next hours and this evening and add some screenshots to Issue https://github.com/maxieds/ChameleonMiniLiveDebugger/issues/4

maxieds, 2018-01-15:
Ok, I'm working on the last bit of uploading a downloaded card image via XModem. I should have the results later today.

maxieds, 2018-01-15:
**v0.2.2 on Play Store (Release Notes):** Incremental updates. Everything but the uploading of a card via XModem works well now. See the [source release](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.2) on the project page for more details.

maxieds, 2018-01-15:
@gtpy: How are you installing the sources? Are you going through Play Store or using the signed APK's from the GitHub repo releases? Just curious. I'm not seeing a lot of traffic through the Play Store pages.

gtpy, 2018-01-15:
@maxieds  I installed first via PlayStore, but are registred as beta tester.
v0.2.1I installed the signed APK because it wasn't accessible at this time in PlayStore.

In the evening I'll check out the up- and download features of your app.

maxieds, 2018-01-15:
The uploading part is a little tricky. Would you check the source repo to
see if you can get it to work?

Right now the transfer in ExportTools freezes the UI after I send the
BYTE_EOF frame to the device. I'm using the same images I downloaded from
the device for testing, so that shouldn't be the issue. If I instead send
BYTE_EOT the GUI doesn't freeze up, but no change is made to the card slot.
This is mostly working, but I have some other things to get done today so I
can't finish it right now. I'm working from the XModem.c logic in the CM
source. Weird.

maxieds, 2018-01-15:
#### RE: Hacking the source code: 
In particular, the code is still there in the latest release, but it's shortcut out with a status message in the actual production APK. The relevant lines / functions to find are these:
* [Line 990 of LiveLoggerActivity.java](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/fff0edbdfba509f476589243006541876170584e/app/src/main/java/com/maxieds/chameleonminilivedebugger/LiveLoggerActivity.java#L990): Calls the relevant initiation code in ExportTools
* [Line 348 of ExportTools.java](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/fff0edbdfba509f476589243006541876170584e/app/src/main/java/com/maxieds/chameleonminilivedebugger/ExportTools.java#L348): Sets up the transfer
* [Line 316 of ExportTools.java](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/fff0edbdfba509f476589243006541876170584e/app/src/main/java/com/maxieds/chameleonminilivedebugger/ExportTools.java#L316): Is what causes the freeze in the UI. Not sure what I should be sending to the device here to indicate that it's the end of the card data transfer.

For reference, I'm looking at [this logic](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Terminal/XModem.c#L75) in the Chameleon Mini source to see what the board is expecting from the XModem transfer. Do you have any suggestions for getting this feature to work?

maxieds, 2018-01-15:
@gtpy 
The latest release (**v0.2.3**) now includes working card uploads via XModem for you to test. I managed to get it working.

gtpy, 2018-01-15:
Nice work @maxieds 
I will work with it this evening and report to you.

maxieds, 2018-01-15:
I'm not sure that there is a repo of card binaries for testing, so here is one I created with a random UID for testing purposes: 
[carddata-MF_CLASSIC_4K_7B-2018-01-15-151412.zip](https://github.com/maxieds/ChameleonMiniLiveDebugger/files/1632905/carddata-MF_CLASSIC_4K_7B-2018-01-15-151412.zip).

I'm sure this has been asked before as well, and I know that the non-NDA docs for this application are scarce, but why is there not even a decent implementation of a Desfire card for the Chameleon Mini? I could probably take a while a write a decent approximation to one if I didn't care about whether the layout of my files necessarily matched the official specs for the card -- as long as to a reader the Chameleon seems Desfire-enough to pass for one. Surely I'm in for a lecture here?

maxieds, 2018-01-15:
Also, I don't want to be the first one to [rate my app](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) on the Play Store. @gtpy: Once you're done testing, if you like it enough, would you write an official review?

maxieds, 2018-01-16:
Any bugs found yet? The [most recent version (v0.2.5-release)](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/current-apk) not quite, but almost all on Play Store _* should be stable for a while *_. I added a number picker for the dip settings and alphabetized the advanced commands spinner for consistency. Also updated the anchor style in the About dialog since the default blue links were a little too ugly for the stylized app. Other than that I've done most of the testing of features I can do for now. I need some input from other testers on other Droid platforms to see if it's all working tightly.

gtpy, 2018-01-16:
@maxieds 
Looks really nice.
Sorry couldn’t more test yesterday. My kids was ill so I had other preferences.
But today at work I take a look at the new version.
You add really fast new features :-) so I need to be faster with testing.

I’ll write the first official review in Play Store - before I can do it I must leave the beta program to rate officially.
I see in Play Store only v0.2.4

maxieds, 2018-01-16:
Version v0.2.5 will be rolled out in the Play Store soon. Thanks for testing.

maxieds, 2018-01-16:
Thanks for the kind review on play store. How's the testing going? Any bugs
I need to fix?

gtpy, 2018-01-17:
Actually no problems.
Sometimes it freezes for some seconds but I’m still testing in which constellations.
If I finished my tests I’ll send you a report about it.

Cheers

maxieds, 2018-01-17:
Any idea when the link will get posted to the getting started page? I think
that would greatly increase the user base.

Maxie

maxieds, 2018-01-17:
Also, what do you think about adding themes? The green is nice enough I
guess.

gtpy, 2018-01-17:
I think green is enough. Not really needed. If all is running perfect you can maybe add some themes.

-------- Original-Nachricht --------

maxieds, 2018-01-17:
I think [this release (v0.2.6)[https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/current-apk] should fix some of the freezing errors. There was a small bug when trying to acquire the serial port lock when doing the top window status updates concurrently with the Export tab features. Seems more stable now. 

Any word on the link, or do you want to test the app more thoroughly first?

gtpy, 2018-01-17:
I think you already seen the answer. https://github.com/emsec/ChameleonMini/issues/159 It’ll be added there.

maxieds, 2018-01-17:
Thank you. I have an algebra comp exam in about 10 hours that I really need to tend to. Wish me luck!

gtpy, 2018-01-17:
I'm sure you'll do it.

-------- Original-Nachricht --------

maxieds, 2018-01-17:
As it turns out, it's probably going to be cancelled today due to light snow. The city of Atlanta is completely unprepared for 2 inches of snow. 

I'm officially going to close this thread since it seems like we've resolved the testing part of this for now. Please do add your screenshots to the other thread when you have a chance.

closed 2018-01-17 by maxieds

maxieds, 2018-01-17:
@gtpy: *RE: Themes* 
I think this is a lot simpler than you thought it was going to be. All I had to do was add flavors to the gradle file and redefine some color combinations. See [these themes](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.7) and the next issue for screenshots.