maxieds/ChameleonMiniLiveDebugger #7: Themes

maxieds, 2018-01-17:
There are now themed versions of the app available. The functionality is identical, but now the user can pick their favorite color combination styling from the [repo APK's](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.7). Note that the Play Store version will remain just the stock green "main" release. Screenshots below:
[image]
[image]
[image]
[image]

maxieds, 2018-01-17:
Each flavor has its own icon:
[image]

maxieds, 2018-01-21:
### "Paid" flavor on Play Store now wraps up the themed UI features:
I decided to create a ["paid" flavor]([Play Store](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger.paid) of the application for those who want to support the project, but do not have the resources to contribute pricey hardware to the development process ($5 for themes). An example "Redmond" (someone besides me should be laughing at this, :dollar: :euro:) theme is featured in the following screenshots for reference:

[image]
[image]

I might also add some testing Desfire card implementations in **res/raw** to the "paid" flavor at some point to incentivize more people who actually use the application to support the development process. Free license, not endless developer resources to contribute kind of thing. At any rate, enjoy the mock theme. The same new release (v0.2.9-free) is available [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.9) for testing.

gtpy, 2018-01-21:
I honor your hard work in this project, but the free version and paid should have same functionality.
More flavors for the paid are ok, but if you cut functions it is not the thinking of an open source project. I like contributing  with testing, coding, analyzing projects but only to help the whole community. Donations are voluntary if someone like a project and will help the developers.
(do not take it personally, it is only a slight criticism)

maxieds, 2018-01-21:
So perhaps the [Desfire testing](https://github.com/maxieds/ChameleonMiniLiveDebugger/wiki/Desfire_card_support) should go into all releases then? It's kind of on the back burner anyways.

As for the current differences between the "paid" and "free" flavors: there essentially are none., The paid version has an extra menu when the user can pick their favorite theme (green gets a little ho hum, which is why I wanted to add the themes to begin with). I think this is fair. It's not Chameleon-specific functionality that is left out. If a user of the app wants to contribute to my semi-lucrative (I sincerely jest here) side career as a software engineer (current [math grad student](http://people.math.gatech.edu/~mschmidt34/)), or just cannot live without having the color options, then they can offer up the $5 for the "paid" flavor. You have to put something extra, however novel, into the paid release so it's different somehow. This is what I was aiming for. Any improvements to the core of the program will of course be available in the current production releases of both builds. There's always the option to roll-your-own release from source with Android studio ...

gtpy, 2018-01-21:
I understand your view. 
It is up to you what you do with your work :-)
I only wanted to explain my own conception-

maxieds, 2018-01-21:
BTW, the reason I didn't roll out the [v0.2.9 release](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.9) to Play Store in the free version is that I didn't want to risk breaking anything with the theme handling in the known stable release until I have more time to use it and test it (it was involved to get this functioning correctly believe it or not). For now the v0.2.8-free version on Play Store is known to work well, so I'm leaving it. If you crave the extra three buttons in the Tools Menu UI, you can grab the pre-release APK off the project website and install it manually.

maxieds, 2018-01-21:
@gtpy: Would you email me separately from this thread. I'd like to send you a copy of the "paid" flavor APK for all of your devoted hard work testing the app, but I don't believe I actually have your GMail.

maxieds, 2018-01-21:
Are there any (a subset of) themes that you want to see included in the free release? That way at least both versions have the same functionality up to data sets. If you want to modify my [style file](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/master/app/src/main/res/values/styles.xml) and send me a few themes for the free release versions, I'd be happy to include them in both. [_Missing the standard linux terminal green on black look notably_]

I used [this reference](https://material.io/guidelines/style/color.html#color-color-palette) (and [here](https://highintegritydesign.com/tools/tinter-shader/)) for most of the themes included in the "paid" flavor released this morning anyhow. There's always the option to just generate a reasonable theme given a single hex code as well.

maxieds, 2018-01-21:
The latest release on the repo site includes three themes for the "free" flavor: Standard Green, Chicky (my favorite and for beta testers), and LinuxGreenOnBlack. I'm still holding back on rolling out the free version with themes to production on Play Store. Give it a few days.

gtpy, 2018-01-23:
Latest free release runs great. Yet doesn't found big bugs. If I have more free time between real life I can debug deeper.

closed 2018-05-07 by maxieds