maxieds/ChameleonMiniLiveDebugger #8: Need better documentation for the app

maxieds, 2018-01-20:
Most of the functionality is pretty self-explanatory *if and only if* you are already very familiar with [this page](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__command_line.html) on the Chameleon command structure. Since the links to my app were posted on the Chameleon [Getting Started](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__getting_started.html) page, I've gotten a couple of confused emails about installation / how to compile too.  

Also, are there any other suggestions for other countries to try rollouts to on Play Store? I almost hesitate to ask this as much as I hesitate to go through and click the buttons to rollout to the 100+ choices available from the Play Store console (some cannot be selected due to US export restrictions hoewever)? Do you all think it's enough to just have it where it is now in the Play Store and link to the most recent APK's from the [releases page](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases)?

maxieds, 2018-01-21:
Here's a [start](https://github.com/maxieds/ChameleonMiniLiveDebugger/wiki/Documentation).

gtpy, 2018-01-23:
If you need some support for documentation, you need to give me the authorization and I can help a bit.

maxieds, 2018-01-26:
@gtpy: I just allowed public editing of the wiki for the time being. Please try to collaborate on [this page](https://github.com/maxieds/ChameleonMiniLiveDebugger/wiki/Documentation).

closed 2018-05-07 by maxieds

maxieds, 2020-08-18:
I am going to give the [WIKI page](https://github.com/maxieds/ChameleonMiniLiveDebugger/wiki) an overhaul in the name of recent documentation that I have neglected properly so far. Please excuse any further posts to this thread for right now, as I need a place to easily upload images and screenshots to make this happen.

maxieds, 2020-08-18:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

maxieds, 2020-08-18:
[image]