maxieds/ChameleonMiniLiveDebugger #11: French Translations

maxieds, 2018-05-12:
As suggested by @iceman1001 [here](https://github.com/iceman1001/ChameleonMini-rebooted/issues/37). Please continue this discussion here.

kevin2008-01, 2018-05-12:
i'm ready... see your instruction for translate.

maxieds, 2018-05-12:
Ok, I guess a good place to start would be [this README](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/master/README.md). After that, the user documentation [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/wiki/Documentation) would be nice. I will add separate files for each of these in the repo/wiki, respectively. 

It never really occurred to me that this could be a big difference maker (of course the boards are manufactured overseas from my perspective in either case). Thanks.

kevin2008-01, 2018-05-12:
i see... you are working man... now the time, is for the world.... for me, the technical french, Italian, dutch, translation is into, my head......0101000010010.... give me the file
for translation...+++

kevin2008-01, 2018-05-12:
[image]
[image]

maxieds, 2018-05-12:
Can you try to do some testing by changing configurations (buttons at the
top of the "Tools" tab) and with the RevE button functionality at the
bottom of the "Tools" tab?

kevin2008-01, 2018-05-12:
sure, now i'm go to sleeping. see you later.

kevin2008-01, 2018-05-13:
Now when i connect the mini chameleon rebooted, the app crash. See the screen.
[image]

maxieds, 2018-05-14:
I'm going to have to figure this out when I get my RevE board. Should be
later this week or early next week.

gtpy, 2018-05-14:
Same if I connect my rev.G board with the app.
Didn't use it with the app a longer time but now it crashes with the white screen

-------- Original-Nachricht --------

maxieds, 2018-05-14:
> maxieds/ChameleonMiniLiveDebugger/issues/11#issuecomment-388862448), or
> ANOycVK2LF1vr9EwdUPrPoRaQ-nFxGK5ks5tyaUCgaJpZM4T8iil).

Again, my board is broken so I'm going to have to fix this when I get my
new RevG in the mail. Should be in the next couple of days (just departed
NY for ATL this morning according to USPS). Sorry for the interruption in
service I can send you the v0.3.6-free APKs if you want to reinstall that
version in the meantime (attached).

gtpy, 2018-05-15:
No Problem. Only want to Report it to you. No hurry with it.

-------- Original-Nachricht --------

maxieds, 2018-05-19:
Could you recheck that the app now works with the most recent (0.4.6-free) release of the app. It should be getting rolled out to Play Store momentarily. It turned out to be a bug with setting the timeout values on the board (returning a 202:INVALID CODE string) which caused an exception to be thrown by the app when parsing some commands. The default exception handler (which I removed) was keeping the app alive with the white screen.

maxieds, 2018-05-19:
@kevin2008-01 : Can you test out the [new version](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/latest) of the app (and on Play Store) on the RevE boards and let me know if it is working now?

maxieds, 2018-05-22:
See #12 for reference. My RevE finally arrived in the mail!

kevin2008-01, 2018-05-22:
I'm on hollidays i'm sorry 

--------------------------------------------
En date de : Mar 22.5.18, Maxie D. Schmidt <notifications@github.com> a écrit :

 Objet: Re: [maxieds/ChameleonMiniLiveDebugger] French Translations (#11)
 À: "maxieds/ChameleonMiniLiveDebugger" <ChameleonMiniLiveDebugger@noreply.github.com>
 Cc: "kevin2008-01" <dylandu54@yahoo.fr>, "Mention" <mention@noreply.github.com>
 Date: Mardi 22 mai 2018, 13h16
 
 See #12
 for reference. My RevE finally arrived in the mail!
 
 —
 You are
 receiving this because you were mentioned.
 Reply to
 this email directly, view
 it on GitHub, or mute
 the thread.

kevin2008-01, 2018-05-25:
Unknow command 200:
VERSION?
RSSI?
IDENTIFY
Invalid command usage 201
RSSIMY

maxieds, 2018-05-27:
Version, rssi, and identify are RevG commands. I have noticed the problem
with rssimy? Not sure why it does this since the syntax looks correct.

maxieds, 2018-05-27:
Does rssimy? work on a standard serial terminal?

kevin2008-01, 2018-05-28:
Yes it work with terminal
--> RSSIMY?
101:OK WITH TEXT
    0 mV