maxieds/ChameleonMiniLiveDebugger #12: The app now works with RevE Boards!

maxieds, 2018-05-22:
Proof that the app now works with the RevE (rebooted) boards:
[image]

fmauNeko, 2019-07-23:
This breaks when using version 1.3 of RevE Rebooted, as the 'MY' command suffix has been removed in commit https://github.com/iceman1001/ChameleonMini-rebooted/commit/6b0f4c9f3339d70d9e814eec3f046e55425cbbf5.

maxieds, 2019-07-24:
@fmauNeko Let me work on fixing this issue. Why Iceman's latest release would intentionally break something like this bothers me. I will post a new release here and on Play Store once I'm done.

maxieds, 2019-08-07:
@iceman1001
Can you please confirm that all recent versions of the RevE firmware will from this point on NOT use the "my" suffix? The [terminal docs](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Terminal-Commands) still suggests that this convention is currently employed. I want to change the CMLD APK only if it benefits most users.

maxieds, 2019-08-07:
@fmauNeko
Please verify that the intended behavior checks out with the attached APK... 

[app-free-release-revefix.apk.zip](https://github.com/maxieds/ChameleonMiniLiveDebugger/files/3479170/app-free-release-revefix.apk.zip)

iceman1001, 2019-08-12:
Yup,  We decided to remove that MY extension,   its been in our plan since we saw that they modified it.
Its a confusing addition that didn't serve any users in the end.   The GUI supports both style,  since the RevE's are sold with old firmware which still has MY-commands.   So your debugger,  should handle RevG / RevE with normal commands,  and RevE with MY-commands.

closed 2020-06-28 by maxieds