maxieds/ChameleonMiniLiveDebugger #14: Not compatible with Samsung S7 Edge (?)

NeuerUser, 2018-06-13:
I cannot install from the Play Store as the store claims that the app is "not compatible" with my phone.
What is the reason? Error on the declaration or really not compatible?

maxieds, 2018-06-13:
What version of Android are you running?

NeuerUser, 2018-06-13:
Same more info:
Samsung SM-935F (European Samsung Galaxy S7 Edge), with official oreo 8.0 firmware.

maxieds, 2018-06-13:
I'm running Oreo 8.1 on my phone and it works fine. I think that the compatibility should be based on the Android API, which must be minimally set to 21. If this is the case your phone should be supported. Can you try installing one of the signed APKs I posted to the repo [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/latest)? It's the same version uploaded to Play Store without Google repackaging / signing it.

NeuerUser, 2018-06-13:
OK, will try and report back. Thx!

NeuerUser, 2018-06-13:
I installed the APK from the repo and it does install and run fine. I, however, do not have yet a Chameleon (it's ordered but not yet delivered), so I cannot test the communication with the device itself. But otherwise the app seems to work without any problems (FCs or other irregularities). So, why does Play store not allow the installation? Any idea?

maxieds, 2018-06-13:
I'm not sure how they determine device compatibility. You might try filing
a bug report so it will let you install from the Play Store.

NeuerUser, 2018-06-13:
Hmm, honestly, I have no idea where to file a bug report with Google... They hide that pretty well... :(

gn0st1c, 2018-10-26:
same here, it says not compatible with any of my devices. installed the apk to my Samsung Note 9 (Oreo 8.1.0) and it worked fine.

[image]

maxieds, 2018-10-28:
> i found out why it was "not compatible" with my devices. it says "This
> item isn't available in your country."
>
>
> https://support.google.com/googleplay/android-developer/answer/7550024?hl=en

Which country are you in? I can add it to the list. The problem is that
play store makes you enable countries one by one from a list of many...

mathdo, 2018-11-26:
Could you Plesse add Germany to the supported countries list (vor Europe if possible)?
Thanks 😄

maxieds, 2018-11-26:
Germany is already on the supported countries list. Are you still having
trouble installing from there?

mathdo, 2018-11-27:
Oh ok, well I cannot find the app in Google Play. But using the google play website I get the infromation that my devices are incompatible (I guess that's also the reason why I cannot find it directly at Google Play!?). The devices I tried are a Sony Z5 compact (running Andorid 7.1.1) and an older Nexus 7 tablet (flo, running LineageOS Android 8). Both are running Android with German language, maybe I try to switch to English and see if that changes something...

scolytus, 2019-01-17:
@maxieds I also see it as incompatible:

Austria, language German
Android 9 on Nokia 6.1

maxieds, 2020-06-28:
This has been a while since there have been updates. Closing issue as not relevant any more.

scolytus, 2020-07-01:
@maxieds But I still see it as not compatible?

Austria, language German,
Android 10 on Nokia 6.1

maxieds, 2020-07-02:
@scolytus 
Can you verify that your phone has NFC and bluetooth permissions available?

maxieds, 2020-07-03:
@scolytus 
The Nokia 6.1 device should be compatible with the latest version **v1.0.8-free** rolled out on Play Store this morning. Please check to see if your device now works.

[image]

maxieds, 2020-07-04:
:fireworks: :sparkler: :fireworks: :sparkler: :fireworks: :sparkler:
@scolytus 
In honor of the holiday (in the US), I finally looked into clicking one-by-one all of the country enable locations for the *Chameleon Mini Live Debugger* application in Play Store (in *Free* and *Paid*) flavors. Happy hacker software is free for us all on this day! It's also nonetheless appropriate that the free software is liberated on an Independence Day. I hope you all enjoy the code. 
:fireworks: :sparkler: :fireworks: :sparkler: :fireworks: :sparkler:

maxieds, 2020-07-04:
The latest release *v1.1.1* should me available within a few hours. It includes bug fixes and some new logging options that are useful when doing Chameleon firmware development. Check out the [releases page](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/latest) for more information.

scolytus, 2020-07-05:
@maxieds I can confirm that I can install both versions now. Thank you!

maxieds, 2020-08-18:
@scolytus 
Issue #26 now makes this a moot point. Ping me back if install issues persist. (Sorry)

closed 2020-08-18 by maxieds