maxieds/ChameleonMiniLiveDebugger #15: Error: Cannot run command since USB is not configured.

TITANForze3D, 2018-06-17:
Hello,
Hardware: Chinese Chameleon Mini ReVE
Firmware: new, upgrade
Apk: ChameleonMiniLiveDebugger v0.5.2-free
Mobile: SM-G531H
Android: 5.1.1
Root ON
Terminal: working
what could be the problem. Thank you

maxieds, 2018-06-17:
Try unplugging the device, closing the app, and then plug in the device
again. Android should start the app for you. Are you running the latest
version of the app? If not you should upgrade.

TITANForze3D, 2018-06-18:
Hello friend
Turned off the phone and turned on with the chameleonemini connected
the phone does not turn on: **Custom binare blocked FRP Lock**
OMG )))

maxieds, 2018-06-19:
???
I meant to turn off the app, not the phone.

TITANForze3D, 2018-06-20:
The application was disabled and turned on, it did not work. Then I tried to restart the phone and it turned into a brick)))

I had to reflash the phone) I put android 8. Today I'll try and unsubscribe.

closed 2018-06-30 by maxieds