maxieds/ChameleonMiniLiveDebugger #16: USB device not found

procrash, 2018-08-21:
Hi,

I have a OnePlus 3+ and the USB Device is not found if I connect the chameleon mini to my android phone. Is there anything I need to consider to get the app to work?

Cheers,

Wolfgang

maxieds, 2018-08-21:
Try disconnecting the device and closing the app. Then reconnect the
device. It should bring up a list of programs and this app should be one of
them. Does your phone have usbc? If not, some older devices will not power
external USB devices, so it could be an issue with your phone as well. Let
me know if this works for you.

maxieds, 2020-06-28:
This has been a while since there have been updates. Closing issue as not relevant any more.

closed 2020-06-28 by maxieds