maxieds/ChameleonMiniLiveDebugger #21: Not compatible with many phones

selxoz, 2019-10-19:
Have tried to install it on Huawei ,HTC, Nexus and others, always says that this app is not compatible.

maxieds, 2019-10-21:
@selxoz  It might be a target or minSDK setting issue in the APK distributed on PlayStore. Google is actually pushing developers to upgrade these even further to make them compatible with only the most recent phones. How old are the phones you tried installing on, and how recently have you patched your Android OS on these phones? 

If all else fails, I always post a signed APK on the [releases page](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/latest) that you can install manually. Just because PlayStore tells you it's not compatible doesn't mean the app will not run if you install it manually this way. Let me know if either of these methods works?

selxoz, 2019-11-03:
Managed to install manually.

closed 2019-11-03 by selxoz