maxieds/ChameleonMiniLiveDebugger #27: CreateProcess error=2

doublemarkpro, 2020-08-16:
I downloaded the project and load it to Android Studio. It says CreateProcess error=2 after Sync Project with Gradle Files. It hints that code:
```
def getShellCommandOutput(String shellCmd) {
    def cmdResult = ['bash', '-c', shellCmd].execute();
    cmdResult.waitFor();
    return cmdResult.text[0..-2];
}
``` has the error: _CreateProcess error=2_

Is there any solution for this error message?

maxieds, 2020-08-18:
Ok. My advice is that whenever you update ``gradle``, there it is typically to be expected that new errors shall arise. I am going to recommend the following course of actions. It is entirely possible that I haven't uploaded enough of the metafiles for a valid Android Studio project. In this case, please work with me to figure out a sane set of instructions to document this procedure. Thank you for supporting the roll-your-own solution of the app!

**Other notes:** Sometimes, running AS on Mac, it takes a little bit of time to update the app's ``build.gradle`` file when you have just recently restarted AS. I think (I will interpret what it is doing) as not fully building the project until after it checks for updates. This is definitely a bug in AS. If it persists with you, please go forth and report it upstream to developers.

### 1. Invalidate caches (File Menu), then restart Android Studio

[image]

### 2. Clean project

[image]

### 3. Make / rebuild project

[image]

maxieds, 2020-08-18:
@doublemarkpro 
Does that fix the issue you are having?

maxieds, 2020-08-18:
@doublemarkpro 
After reviewing [this question](https://stackoverflow.com/questions/27246491/android-studio-gradle-createprocess-error-2/38410801), it behoves me to ask whether you are running AS on Windows, or on an otherwise sane Unix environment (yes, Mac included here). If you are on windows, you are going to want to make sure there is actually a command named ``bash`` and ``git`` in the system(32?) binaries directory. GitHub had a nice installer the last time I checked to get ``git`` installed on Windows (also, a better shell replacement with auto-completion than I found elsewhere). To get bash, maybe you can try the [method here](https://itsfoss.com/install-bash-on-windows/)?

maxieds, 2022-06-16:
I am going to assume that the issue resolved itself by now.

closed 2022-06-16 by maxieds