maxieds/ChameleonMiniLiveDebugger #30: Crash Report: CMLD 1.2.9-free running  P615XXU3BTI3 [SDK 29] on device samsung gta4xleea SM-P615

BeardyMcNormous, 2020-11-03:
@maxieds:

**Crash data summary:**
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
* *[Ljava.lang.String;@94825bf*: [Ljava.lang.String;@ff2748c
**Stack trace:**
" + "```java
ScrollView can host only one direct child

L01| android.widget.ScrollView.
     addView( ScrollView.java:581 )
L02| com.maxieds.chameleonminilivedebugger.b0.
     g( UITabUtils.java:34 )
L03| com.maxieds.chameleonminilivedebugger.b0.
     f( UITabUtils.java:3 )
L04| com.maxieds.chameleonminilivedebugger.x$a.
     b( TabFragment.java:10 )
L05| com.maxieds.chameleonminilivedebugger.x$a.
     f( TabFragment.java:6 )
L06| com.maxieds.chameleonminilivedebugger.x$a$b.
     onClick( TabFragment.java:1 )
L07| android.view.View.
     performClick( View.java:7862 )
L08| android.widget.TextView.
     performClick( TextView.java:15004 )
L09| android.view.View.
     performClickInternal( View.java:7831 )
L10| android.view.View.access$3600( View.java:879 )
L11| android.view.View$PerformClick.
     run( View.java:29359 )
L12| android.os.Handler.
     handleCallback( Handler.java:883 )
L13| android.os.Handler.
     dispatchMessage( Handler.java:100 )
L14| android.os.Looper.
     loop( Looper.java:237 )
L15| android.app.ActivityThread.
     main( ActivityThread.java:8167 )
L16| java.lang.reflect.Method.invoke( Native Method )
L17| com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:496 )
L18| com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:1100 )

```
**Additional comments:**

*NONE*

maxieds, 2020-11-04:
@BeardyMcNormous 
Thank you for being the first one to invoke my uncaught exception handler Activity. It looks like some of the output about the Android hardware was truncated. Did the send to GitHub button work, or did you need to copy and paste the output? Just wondering so I can make some minor improvements if needed.

I think this problem has actually been noticed and fixed in more recent commits over the past couple of weeks. I still haven't pushed a new release onto Play Store because I am in the process of finalizing some new experimental scripting features. Is that what you were looking at when the error messages were displayed? Also, just to check, did you install the application from Play Store?

maxieds, 2021-04-29:
@BeardyMcNormous 
This bug report does not contain enough useful information to keep open. Please post new ones as you encounter them coming in. Thanks. --MDS

closed 2021-04-29 by maxieds