maxieds/ChameleonMiniLiveDebugger #33: Crash Report: CMLD 1.3.2-free running samsung/beyond2qltesq/beyond2q:10/QP1A.190711.020/G975USQU4ETH7:user/release-keys G975USQS4ETK2 [SDK 29] on device samsung beyond2qltesq SM-G975U

SHMAUS-Carter, 2021-01-12:
@maxieds:

**Crash data summary:**
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
* *[Ljava.lang.String;@aa9fd38*: [Ljava.lang.String;@6d81111
**Stack trace:**
" + "```java
Attempt to invoke virtual method 'void android.widget.Button.setOnClickListener( android.view.View$OnClickListener )' on a null object reference

L01| com.maxieds.chameleonminilivedebugger.v.q.
     c( ScriptingGUIMain.java:5 )
L02| com.maxieds.chameleonminilivedebugger.c0.
     f( UITabUtils.java:1 )
L03| com.maxieds.chameleonminilivedebugger.c0.
     g( UITabUtils.java:2 )
L04| com.maxieds.chameleonminilivedebugger.y.
     f0( TabFragment.java:10 )
L05| androidx.fragment.app.Fragment.
     I0( Fragment.java:4 )
L06| androidx.fragment.app.j.
     R0( FragmentManagerImpl.java:70 )
L07| androidx.fragment.app.j.
     O0( FragmentManagerImpl.java:9 )
L08| androidx.fragment.app.j.
     P0( FragmentManagerImpl.java:7 )
L09| androidx.fragment.app.a.
     r( BackStackRecord.java:25 )
L10| androidx.fragment.app.j.
     n0( FragmentManagerImpl.java:6 )
L11| androidx.fragment.app.j.
     o0( FragmentManagerImpl.java:14 )
L12| androidx.fragment.app.j.
     Z0( FragmentManagerImpl.java:10 )
L13| androidx.fragment.app.j.
     m0( FragmentManagerImpl.java:5 )
L14| androidx.fragment.app.a.
     h( BackStackRecord.java:2 )
L15| androidx.fragment.app.m.
     b( FragmentPagerAdapter.java:2 )
L16| androidx.viewpager.widget.ViewPager.
     F( ViewPager.java:50 )
L17| androidx.viewpager.widget.ViewPager.
     E( ViewPager.java:1 )
L18| androidx.viewpager.widget.ViewPager.
     onMeasure( ViewPager.java:25 )
L19| android.view.View.
     measure( View.java:26411 )
L20| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L21| android.widget.LinearLayout.
     measureChildBeforeLayout( LinearLayout.java:1552 )
L22| android.widget.LinearLayout.
     measureVertical( LinearLayout.java:842 )
L23| android.widget.LinearLayout.
     onMeasure( LinearLayout.java:721 )
L24| android.view.View.
     measure( View.java:26411 )
L25| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L26| android.widget.LinearLayout.
     measureChildBeforeLayout( LinearLayout.java:1552 )
L27| android.widget.LinearLayout.
     measureVertical( LinearLayout.java:842 )
L28| android.widget.LinearLayout.
     onMeasure( LinearLayout.java:721 )
L29| android.view.View.
     measure( View.java:26411 )
L30| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L31| android.widget.FrameLayout.
     onMeasure( FrameLayout.java:194 )
L32| androidx.appcompat.widget.ContentFrameLayout.
     onMeasure( ContentFrameLayout.java:21 )
L33| android.view.View.
     measure( View.java:26411 )
L34| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L35| android.widget.LinearLayout.
     measureChildBeforeLayout( LinearLayout.java:1552 )
L36| android.widget.LinearLayout.
     measureVertical( LinearLayout.java:842 )
L37| android.widget.LinearLayout.
     onMeasure( LinearLayout.java:721 )
L38| android.view.View.
     measure( View.java:26411 )
L39| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L40| android.widget.FrameLayout.
     onMeasure( FrameLayout.java:194 )
L41| android.view.View.
     measure( View.java:26411 )
L42| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L43| android.widget.LinearLayout.
     measureChildBeforeLayout( LinearLayout.java:1552 )
L44| android.widget.LinearLayout.
     measureVertical( LinearLayout.java:842 )
L45| android.widget.LinearLayout.
     onMeasure( LinearLayout.java:721 )
L46| android.view.View.
     measure( View.java:26411 )
L47| android.view.ViewGroup.
     measureChildWithMargins( ViewGroup.java:7845 )
L48| android.widget.FrameLayout.
     onMeasure( FrameLayout.java:194 )
L49| com.android.internal.policy.DecorView.
     onMeasure( DecorView.java:1050 )
L50| android.view.View.
     measure( View.java:26411 )
L51| android.view.ViewRootImpl.
     performMeasure( ViewRootImpl.java:3638 )
L52| android.view.ViewRootImpl.
     measureHierarchy( ViewRootImpl.java:2373 )
L53| android.view.ViewRootImpl.
     performTraversals( ViewRootImpl.java:2665 )
L54| android.view.ViewRootImpl.
     doTraversal( ViewRootImpl.java:2225 )
L55| android.view.ViewRootImpl$TraversalRunnable.
     run( ViewRootImpl.java:9126 )
L56| android.view.Choreographer$CallbackRecord.
     run( Choreographer.java:999 )
L57| android.view.Choreographer.
     doCallbacks( Choreographer.java:797 )
L58| android.view.Choreographer.
     doFrame( Choreographer.java:732 )
L59| android.view.Choreographer$FrameDisplayEventReceiver.
     run( Choreographer.java:984 )
L60| android.os.Handler.
     handleCallback( Handler.java:883 )
L61| android.os.Handler.
     dispatchMessage( Handler.java:100 )
L62| android.os.Looper.
     loop( Looper.java:237 )
L63| android.app.ActivityThread.
     main( ActivityThread.java:8167 )
L64| java.lang.reflect.Method.invoke( Native Method )
L65| com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:496 )
L66| com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:1100 )

```
**Additional comments:**

*NONE*

maxieds, 2021-01-12:
@SHMAUS-Carter 
This has been reported to me a couple of times since I released **v1.3.2** early this morning. It doesn't happen on my testing phone. I assume that this error is displayed when you launch the updated application? I am working to fix the error today. Please update the application once a new version is available on Play Store.

maxieds, 2021-01-12:
Thanks to Joar for helping me debug to prepare for the new **v1.3.4** release over email.

@SHMAUS-Carter 
Please upgrade to the [latest release](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/latest) just posted to the Play Store. Can you close out this issue once you verify that the problem is resolved after the upgrade?

SHMAUS-Carter, 2021-01-12:
Sure as soon as I get the updated I'll check to see if it still crashes on launch

SHMAUS-Carter, 2021-01-13:
Fixed

closed 2021-01-13 by SHMAUS-Carter