maxieds/ChameleonMiniLiveDebugger #37: Crash Report (CMLD 1.3.4-free):  Neither user 10278 nor current process has android.permission.BLUETOOTH_ADMIN. [Android 10, SDK 29]

Imminent-Darkness, 2021-01-19:
@maxieds:

**Crash data summary:**
* *CMLD Version*: v1.3.4-free (8176:96)
* *CMLD GitHub Commit*: 1cddba77e675 @ Tue Jan 12 18:18:04 2021 -0500
* *CMLD Build Date*: 2021-01-12-181925
* *Chameleon Type*: <UNKNOWN>
* *Serial Connection Type*: NULL
* *Chameleon Config*: NOT SET
* *Chameleon Logmode*: NOT SET
* *Chameleon Timeout*: NOT SET
* *Android Device*: OnePlus7Pro
* *Android Manufacturer*: OnePlus
* *Android Brand*: OnePlus
* *Android Product*: OnePlus7Pro
* *Android Model*: GM1917
* *Android Hardware*: qcom
* *Android SDK*: 29
* *Android OS Release*:  10 (REL / 1911070200)
* *Android Board*: msmnile
* *Android Type*: user

**Stack trace:**
```java**Exception Message:** Need BLUETOOTH ADMIN permission: Neither user 10278 nor current process has android.permission.BLUETOOTH_ADMIN.

Detailed Stack Trace:
java.lang.SecurityException: Need BLUETOOTH ADMIN permission: Neither user 10278 nor current process has android.permission.BLUETOOTH_ADMIN.
	at android.os.Parcel.
     createException( Parcel.java:2071 )
	at android.os.Parcel.
     readException( Parcel.java:2039 )
	at android.os.Parcel.
     readException( Parcel.java:1987 )
	at android.bluetooth.IBluetooth$Stub$Proxy.
     cancelDiscovery( IBluetooth.java:2289 )
	at android.bluetooth.BluetoothAdapter.
     cancelDiscovery( BluetoothAdapter.java:1709 )
	at com.maxieds.chameleonminilivedebugger.e.
     r( BluetoothGattConnector.java:2 )
	at com.maxieds.chameleonminilivedebugger.f.
     s( BluetoothSerialInterface.java:1 )
	at com.maxieds.chameleonminilivedebugger.n.
     d( ChameleonSettings.java:2 )
	at com.maxieds.chameleonminilivedebugger.c0$a.
     onClick( UITabUtils.java:3 )
	at android.view.View.
     performClick( View.java:7201 )
	at android.view.View.
     performClickInternal( View.java:7170 )
	at android.view.View.access$3500( View.java:806 )
	at android.view.View$PerformClick.
     run( View.java:27562 )
	at android.os.Handler.
     handleCallback( Handler.java:883 )
	at android.os.Handler.
     dispatchMessage( Handler.java:100 )
	at android.os.Looper.
     loop( Looper.java:214 )
	at android.app.ActivityThread.
     main( ActivityThread.java:7682 )
	at java.lang.reflect.Method.invoke( Native Method )
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:516 )
	at com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:950 )

```

**Additional comments:**
*NONE -- Please fill in more details about how you generated this error, like what actions you took before this screen is displayed. Did the error happen on launch of the application? Did you click a button or switch tabs? Any extra details about how the error happened are useful to fixing the issue in future releases.*

The first error got thrown when I tried to allow bluetooth connections from the app. Then I tried to connect my Chameleon Mini Rev G with a cable and the same error got thrown. I see someone else had a similar issue.

maxieds, 2021-01-25:
@Imminent-Darkness 
This is the second time this issue has been brought to my attention. I will work to fix it for the next release.

Does (or did it previously) BT connectivity work for you in the app? I was still under the impression that the Proxgrind RevG devices were not working over BT. I have only actually seen them to work over the serial USB connections any way.

maxieds, 2021-04-29:
@Imminent-Darkness 
This issue should be fixed with the latest **v1.3.6** release of the software. Please upgrade and write back if you still have the issue. Thanks. -- MDS

closed 2021-04-29 by maxieds