maxieds/ChameleonMiniLiveDebugger #49: BT connect crash

clavo2, 2021-06-23:
device pairs to phone.  Works over USBOTA.   Crashes at all attempts of a BT connection.  

BT selected as a pairing option.   Tap connect...  Crashes. 

ANDROID Pixel 3
 - Droid Version 11 
 - Chrome, Brave

 - Device: Pixel 3
 - OS: OS v11

maxieds, 2021-09-03:
@clavo2 @ LordFisk @FFX721 @M1rr0rx

**RE: Issues #49; #47; #48; #50; #51**

The [Bluetooth](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/master/app/src/main/java/com/maxieds/chameleonminilivedebugger/BluetoothGattConnector.java) (1) [connectivity](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/master/app/src/main/java/com/maxieds/chameleonminilivedebugger/BluetoothSerialInterface.java) (2) to a certain make/model of the RevG devices manufactured in China is [known not to work](https://github.com/emsec/ChameleonMini/issues/271). This is due to intentional lack of communication and lack of disclosure to the inner workings of the BT interface details by the Proxgrind manufacturers. You can, of course, use their Android app to connect over BT. But, I warn you that the Chinese originated libraries they use internally as a native "packer" for Android OS have a known trojan horse that people get really, **REALLY** defensive about me pointing out online.

I will try my best this week to patch the BT errors by adding ``try/catch`` statements around the code. N.b., I have to go over to the GA Tech health center today to get a COVID-19 test. That and postdoc applications this year have been keeping me away from writing Java code...

Thank you for taking the time to write out a well reasoned, properly punctuated bug report. I am trying my best. 😸

maxieds, 2021-09-03:
@clavo2 @LordFisk @FFX721 @M1rr0rx
In the midst of trying to fix the BT crash this morning, I realized that I am still unable to push commits to this repo from my MacOS command line. The change to require access tokens on GH comes in with other corporate hacking related to former US president Trump's claimed horror movie themed **2021.08.13** re-insurrection attacks against my country. In the event I have to post a new APK to Google Play Store today without being able to release the corresponding source code (forgive me, sincerely), this is the crux of the new theme changes:

**()**
```xml
              <!--
                     Immortalizing the f*cking fascist by name:
                     https://www.youtube.com/playlist?list=PLPV7hiOR4qwwRNWbz0Xac2gvvJgYz0Kbb
               https://github.com/maxieds/ThrowingSomeOldBitsIntoAHoleOnGitHub/tree/master/HiMyNameIsLuckyHiLucky
                     https://github.com/maxieds/GEyesThemes/releases/tag/Codename-BackOrifice53
                -->
                <RadioButton
                    android:id="@+id/themeRadioFckingFascist"
                    android:buttonTint="?colorPrimaryDark"
                    android:layout_width="wrap_content"
                    android:layout_height="wrap_content"
                    android:checked="false"
                    android:drawableLeft="@drawable/chameleonlogo32_jas0n"
                    android:drawablePadding="5dp"
                    android:text="F*cking Fascist" />
```
Screw you [Jas0n](https://github.com/maxieds/ThrowingSomeOldBitsIntoAHoleOnGitHub/blob/master/HiMyNameIsLuckyHiLucky/recent-emails-uiuc-ad.txt). To the rest of you, enjoy the app as usual. I will be back in touch when the new release is available.

maxieds, 2021-09-03:
That is, my distraction at the moment:
```xml
                <!--
                     Q: Why are all good looking men such conceited jackasses !?!
                     A: Pretty men with only small endowments to speak of in this world have never
                        had to work hard enough to get laid (love 'em, and leave 'em; fuck 'em, or fail 'em).
                     Immortalizing the f*cking fascist by name:
                     https://www.youtube.com/playlist?list=PLPV7hiOR4qwwRNWbz0Xac2gvvJgYz0Kbb
                   https://github.com/maxieds/ThrowingSomeOldBitsIntoAHoleOnGitHub/tree/master/HiMyNameIsLuckyHiLucky
                     https://github.com/maxieds/GEyesThemes/releases/tag/Codename-BackOrifice53
                -->
               <RadioButton
                    android:id="@+id/themeRadioFckingFascist"
                    android:buttonTint="?colorPrimaryDark"
                    android:layout_width="wrap_content"
                    android:layout_height="wrap_content"
                    android:checked="false"
                    android:drawableLeft="@drawable/chameleonlogo32_jas0n"
                    android:drawablePadding="5dp"
                    android:text="F*cking Fascist" />
```
[image]

Sorry all, I have an interview and a COVID test to get to before BT gets fixed this afternoon. :)

closed 2022-06-16 by maxieds