maxieds/ChameleonMiniLiveDebugger #54: Crash Report (CMLD 1.3.6-free): onClick attribute defined on view class androidx.appcompat.widget.f [Android 12, SDK 31]

lvandenb, 2021-12-17:
@maxieds:
The ogger used ond Galaxy s20 android 12

**Crash data summary:**
* *CMLD Version*: v1.3.6-free (8178:98)
* *CMLD GitHub Commit*: 9eec3d4bd94b @ Thu Apr 29 13:50:12 2021 -0400
* *CMLD Build Date*: 2021-04-29-135246
* *Chameleon Type*: KAOS RevG Device
* *Serial Connection Type*: USB
* *Chameleon Config*: MF_DESFIRE
* *Chameleon Logmode*: LIVE
* *Chameleon Timeout*: 5000 ms
* *Android Device*: p3s
* *Android Manufacturer*: samsung
* *Android Brand*: samsung
* *Android Product*: p3sxeea
* *Android Model*: SM-G998B
* *Android Hardware*: exynos2100
* *Android SDK*: 31
* *Android OS Release*: samsung/p3sxeea/p3s:12/SP1A.210812.016/G998BXXU3BUKG:user/release-keys 12 (REL / G998BXXS3BUL1)
* *Android Board*: exynos2100
* *Android Type*: user

**Stack trace:**
```java**Exception Message:** Could not find method actionButtonRestorePeripheralDefaults( View ) in a parent or ancestor Context for android:onClick attribute defined on view class androidx.appcompat.widget.f

Detailed Stack Trace:
java.lang.IllegalStateException: Could not find method actionButtonRestorePeripheralDefaults( View ) in a parent or ancestor Context for android:onClick attribute defined on view class androidx.appcompat.widget.f
	at androidx.appcompat.app.i$a.
     a( AppCompatViewInflater.java:10 )
	at androidx.appcompat.app.i$a.
     onClick( AppCompatViewInflater.java:2 )
	at android.view.View.
     performClick( View.java:7792 )
	at android.widget.TextView.
     performClick( TextView.java:16045 )
	at android.view.View.
     performClickInternal( View.java:7769 )
	at android.view.View.access$3800( View.java:910 )
	at android.view.View$PerformClick.
     run( View.java:30184 )
	at android.os.Handler.
     handleCallback( Handler.java:938 )
	at android.os.Handler.
     dispatchMessage( Handler.java:99 )
	at android.os.Looper.
     loopOnce( Looper.java:226 )
	at android.os.Looper.
     loop( Looper.java:313 )
	at android.app.ActivityThread.
     main( ActivityThread.java:8582 )
	at java.lang.reflect.Method.invoke( Native Method )
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:563 )
	at com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:1133 )

```

**Additional comments:**
*NONE -- Please fill in more details about how you generated this error, like what actions you took before this screen is displayed. Did the error happen on launch of the application? Did you click a button or switch tabs? Any extra details about how the error happened are useful to fixing the issue in future releases.*

maxieds, 2021-12-17:
@lvandenb 
Can you give me more information on the sequence of button clicks and/or tab changes that led to this crash? That will help me track it down and fix it in a new release. I have not run into this issue before. It could be the new Android 12 OS.

lvandenb, 2021-12-20:
It seemed the happen the first time after installation.

Also there were no rights for accessing storage. After granting access, the log file is in the downloads.
Still mailing generates a security error.

closed 2022-06-16 by maxieds