maxieds/ChameleonMiniLiveDebugger #58: Crash on Samsung wit Kaspar Oswald's ChameleonMini

annee2000, 2022-05-27:
APP crashes as soon as I connect the PCB.
Samsung S10
Android 12. latest version, March 2022
Chameleonmini, firmware March 2022

**Exception Message:** Unable to start activity ComponentInfo{com.maxieds.chameleonminilivedebugger/com.maxieds.chameleonminilivedebugger.LiveLoggerActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.view.View.setPadding( int, int, int, int )' on a null object reference

Detailed Stack Trace:
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.maxieds.chameleonminilivedebugger/com.maxieds.chameleonminilivedebugger.LiveLoggerActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.view.View.setPadding( int, int, int, int )' on a null object reference
	at android.app.ActivityThread.
     performLaunchActivity( ActivityThread.java:4035 )
	at android.app.ActivityThread.
     handleLaunchActivity( ActivityThread.java:4201 )
	at android.app.servertransaction.LaunchActivityItem.
     execute( LaunchActivityItem.java:103 )
	at android.app.servertransaction.TransactionExecutor.
     executeCallbacks( TransactionExecutor.java:135 )
	at android.app.servertransaction.TransactionExecutor.
     execute( TransactionExecutor.java:95 )
	at android.app.ActivityThread$H.
     handleMessage( ActivityThread.java:2438 )
	at android.os.Handler.
     dispatchMessage( Handler.java:106 )
	at android.os.Looper.
     loopOnce( Looper.java:226 )
	at android.os.Looper.
     loop( Looper.java:313 )
	at android.app.ActivityThread.
     main( ActivityThread.java:8663 )
	at java.lang.reflect.Method.invoke( Native Method )
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:567 )
	at com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:1135 )
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.view.View.setPadding( int, int, int, int )' on a null object reference
	at com.maxieds.chameleonminilivedebugger.d0.
     h( Utils.java:3 )
	at com.maxieds.chameleonminilivedebugger.d0.
     i( Utils.java:1 )
	at com.maxieds.chameleonminilivedebugger.d0.
     j( Utils.java:1 )
	at com.maxieds.chameleonminilivedebugger.LiveLoggerActivity.
     onCreate( LiveLoggerActivity.java:43 )
	at android.app.Activity.
     performCreate( Activity.java:8290 )
	at android.app.Activity.
     performCreate( Activity.java:8270 )
	at android.app.Instrumentation.
     callActivityOnCreate( Instrumentation.java:1329 )
	at android.app.ActivityThread.
     performLaunchActivity( ActivityThread.java:4009 )
	... 12 more

maxieds, 2022-06-02:
@annee2000 
Thank you bringing this to  my attention. I have not tested with Android 12. I suspect that the platform introduced some new bugs (as usual). I only have access to Android 12 with an emulator. As such, I am not able to connect the USB device and generate/test the exception that you found. The latest commit may resolve the issue. Please test with the attached APK (signed by the same developer signature as the Play Store versions). If this fixes the Android 12 issues moving forward, I will push a new release onto the Play Store so it is easier to install:
[ChameleonMiniLiveDebugger-v1.4.0-2022-06-02-031039-free-release.apk.zip](https://github.com/maxieds/ChameleonMiniLiveDebugger/files/8821556/ChameleonMiniLiveDebugger-v1.4.0-2022-06-02-031039-free-release.apk.zip)

annee2000, 2022-06-02:
Hi, thanks will be done Tuesday.
Best
Detlef

Send from my mobile!

maxieds, 2022-06-16:
@annee2000 
There is a new (*v1.4.1-free*) version of CMLD on Google Play Store as of this morning (EST/US time). It should fix your issue based on my testing with the Android 12 emulator. Please reopen the issue if you still experience related crashes after upgrading to the latest release. And similarly for #59.

annee2000, 2022-06-17:
Hi Maxie, sorry for the late reply. The APP still crashes .
I am not familiar on how to reopen the issue on Github.
Best

annee2000, 2022-06-17:
**Exception Message:** Unable to start activity
ComponentInfo{com.maxieds.chameleonminilivedebugger/com.maxieds.chameleonminilivedebugger.LiveLoggerActivity}:
java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN
permission for AttributionSource { uid = 10408, packageName =
com.maxieds.chameleonminilivedebugger, attributionTag = null, token =
***@***.***, next = null }: AdapterService
cancelDiscovery

Detailed Stack Trace:
java.lang.RuntimeException: Unable to start activity
ComponentInfo{com.maxieds.chameleonminilivedebugger/com.maxieds.chameleonminilivedebugger.LiveLoggerActivity}:
java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN
permission for AttributionSource { uid = 10408, packageName =
com.maxieds.chameleonminilivedebugger, attributionTag = null, token =
***@***.***, next = null }: AdapterService
cancelDiscovery
at android.app.ActivityThread.
     performLaunchActivity( ActivityThread.java:4035 )
at android.app.ActivityThread.
     handleLaunchActivity( ActivityThread.java:4201 )
at android.app.servertransaction.LaunchActivityItem.
     execute( LaunchActivityItem.java:103 )
at android.app.servertransaction.TransactionExecutor.
     executeCallbacks( TransactionExecutor.java:135 )
at android.app.servertransaction.TransactionExecutor.
     execute( TransactionExecutor.java:95 )
at android.app.ActivityThread$H.
     handleMessage( ActivityThread.java:2438 )
at android.os.Handler.
     dispatchMessage( Handler.java:106 )
at android.os.Looper.
     loopOnce( Looper.java:226 )
at android.os.Looper.
     loop( Looper.java:313 )
at android.app.ActivityThread.
     main( ActivityThread.java:8663 )
at java.lang.reflect.Method.invoke( Native Method )
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:567 )
at com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:1135 )
Caused by: java.lang.SecurityException: Need
android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid =
10408, packageName = com.maxieds.chameleonminilivedebugger, attributionTag

AdapterService cancelDiscovery
at android.os.Parcel.
     createExceptionOrNull( Parcel.java:2437 )
at android.os.Parcel.
     createException( Parcel.java:2421 )
at android.os.Parcel.
     readException( Parcel.java:2404 )
at android.os.Parcel.
     readException( Parcel.java:2346 )
at android.bluetooth.IBluetooth$Stub$Proxy.
     cancelDiscovery( IBluetooth.java:5036 )
at android.bluetooth.BluetoothAdapter.
     cancelDiscovery( BluetoothAdapter.java:2955 )
at com.maxieds.chameleonminilivedebugger.e.
     r( BluetoothGattConnector.java:2 )
at com.maxieds.chameleonminilivedebugger.f.
     s( BluetoothSerialInterface.java:1 )
at com.maxieds.chameleonminilivedebugger.n.
     d( ChameleonSettings.java:2 )
     onNewIntent( LiveLoggerActivity.java:9 )
     onCreate( LiveLoggerActivity.java:10 )

Send from my mobile!

maxieds, 2022-06-17:
@annee2000 
Please try this updated APK built from the latest commit made earlier today. It may resolve the issue. If not, please post the new stack trace information.

[ChameleonMiniLiveDebugger-v1.4.2-2022-06-17-121749-free-release.apk.zip](https://github.com/maxieds/ChameleonMiniLiveDebugger/files/8929119/ChameleonMiniLiveDebugger-v1.4.2-2022-06-17-121749-free-release.apk.zip)

annee2000, 2022-06-17:
this time no crash at startup. Reading og card was not possible. After a
while the App crashed again
Here we are again with an export:

**Exception Message:** Could not execute method for android:onClick

Detailed Stack Trace:
java.lang.IllegalStateException: Could not execute method for
android:onClick
at c.h$a.
     onClick( AppCompatViewInflater.java:4 )
at android.view.View.
     performClick( View.java:7792 )
at android.widget.TextView.
     performClick( TextView.java:16112 )
at android.view.View.
     performClickInternal( View.java:7769 )
at android.view.View.access$3800( View.java:910 )
at android.view.View$PerformClick.
     run( View.java:30218 )
at android.os.Handler.
     handleCallback( Handler.java:938 )
at android.os.Handler.
     dispatchMessage( Handler.java:99 )
at android.os.Looper.
     loopOnce( Looper.java:226 )
at android.os.Looper.
     loop( Looper.java:313 )
at android.app.ActivityThread.
     main( ActivityThread.java:8663 )
at java.lang.reflect.Method.invoke( Native Method )
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.
     run( RuntimeInit.java:567 )
at com.android.internal.os.ZygoteInit.
     main( ZygoteInit.java:1135 )
Caused by: java.lang.reflect.InvocationTargetException
at java.lang.reflect.Method.invoke( Native Method )
at c.h$a.
     onClick( AppCompatViewInflater.java:3 )
... 13 more
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method
'boolean android.widget.RadioButton.isChecked(  )' on a null object
reference
at com.maxieds.chameleonminilivedebugger.m.
     a( ExternalFileIO.java:19 )
at com.maxieds.chameleonminilivedebugger.LiveLoggerActivity.
     actionButtonWriteFile( LiveLoggerActivity.java:2 )
... 15 more

Send from my mobile!

maxieds, 2022-06-22:
@annee2000 
When you report exceptions from the crash handler, **please** include what you did right before the crash happened (changed tab, clicked a button, etc.). As you can see from the data it currently generates above, the stack trace information needs to get better. I added [this project](https://github.com/maxieds/ChameleonMiniLiveDebugger/projects/11) -- no idea when it will get finished. 

I am doing some testing with a new phone tonight. I will see if I can get the BT connectivity better debugged.

annee2000, 2022-06-22:
Hi Maxie,
I did not use the BT connection at all. From what I recall I:
Startet caméléon App
Slot 1 showed active
I selected reader
Wanted to read a card for test
Did not work
Disconnected device
Reconnectet device and all slots were red (disconnected)
Tried to manually select 1st slot, not working
Disconnected device an reconnected
APP crashed - I send the error log

Hope this helps. APP worked fine on my old Motorola with Android 7
Best Detlef

Send from my mobile!

maxieds, 2022-06-23:
Here is an APK built with the latest commit that should be more cautious handling GUI ``View`` object handles obtained by their resource ID. In short, I expect it to crash less frequently on Android phones that I do not have hardware to test on. I am still testing some new code to work with BT devices. Please take some time to test with the latest and see if the crashes still happen. Once things appear stable, I will push an official release to Play Store for convenient reinstallation. 

[ChameleonMiniLiveDebugger-v1.4.3-beta-2022-06-22-223802-free-release.apk.zip](https://github.com/maxieds/ChameleonMiniLiveDebugger/files/8963197/ChameleonMiniLiveDebugger-v1.4.3-beta-2022-06-22-223802-free-release.apk.zip)

maxieds, 2022-07-07:
A new version (**v1.4.3-free**) will be on Play Store soon. Please upgrade and reopen the issue if problems persist.

closed 2022-07-07 by maxieds