maxieds/ChameleonMiniLiveDebugger #60: UPDATE: Support for Chameleon connections using Bluetooth BLE is under active development

maxieds, 2022-07-22:
Recent [releases](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases) of this application (CMLD) have fixed Android permissions errors for the optional configuration option to enable Bluetooth (BT) connections (checkbox under the ``Settings > General`` tab). I am working towards getting the BT connections for the Proxgrind RevG / Tiny / TinyPro Chameleon devices. I will be testing this code soon and hope to rollout a new release that adds BT support soon (fingers crossed).

In the meantime, I would like to gather some information from helpful users. If you use the Chameleon Mini over a Bluetooth connection, can you please report any alternate names/aliases (things that your device is called by the BT stack on your system of choice)? I have observed the following three so far. Newer releases of the onboard BT firmware may have changed things since the last time I looked at this:

* *BLE-Chameleon*
* *Chameleon*
* *ChameleonTiny*

maxieds, 2022-08-13:
Note that [this project](https://github.com/maxieds/ChameleonMiniLiveDebugger/projects/5) to get BLE/Bluetooth Chameleon device connections working is still in progress. More commits with progress and further testing towards this goal will be posted soon. Keep an eye on the [most recent pre-releases](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases) for signed beta testing APKs before this functionality is pushed to Google Play Store. **Beta testers are very much appreciated!**

Msprg, 2022-11-05:
Oh, hello there!

I just had issues connecting my rev.G mini via BLE and was about to ask you about troubleshooting - but luckily I decided to look here into the issues first.

Well... suppose I'm a beta tester now 😅
I'm running emsec firmware. If I were to switch to the rrg firmware, I could connect and use their app just fine, so Bluetooth itself works.

About the Bluetooth firmware upgrade that is mentioned in your in-app troubleshooting steps: Is it the thing where you in proxgrind app tap something like "BLE CMD version" like 4 times and a "secret" update dialog comes up? Or is it something else you were mentioning?

If you need me to try something specific, I'd be happy to assist - I already have to thank you for tirelessly implementing the desfire emulation! Now that I think about it, I may actually even owe you 😁

Have a nice day~!

ShigemoriHakura, 2023-02-09:
I accidently get my chameleon mini's bluetooth firmware lost, any chance I can get a firmware to flash in?

maxieds, 2025-03-21:
It may be the case that the RFID group's interface is [described here](https://github.com/RfidResearchGroup/ChameleonMini/blob/proxgrind/Firmware/Chameleon-Mini/uartcmd.h). I will not be supporting this app for a while.